the-raspberry-pi-guy/lcd #1: install.sh

Loki69, 2017-03-12:
on 25 line you do "cp installConfigs/modules /etc/ "
but in modules have only one new string i2c-dev
mb.  write echo "i2c-dev" >> /etc/modules ?

amzaat, 2019-12-26:
the witclone is not available on githonb, I bought your book learn robotics with raspberry pi, but no link on GitHub for the wii, I bought especially for this purpose

the-raspberry-pi-guy, 2019-12-26:
Hi Amzaat, thanks for buying a copy of my book. This is another one of my GitHub repo's, perhaps you meant to open an issue on /raspirobots? Either way, I am not sure I understand the problem you have?

Have you followed the instructions in the book for downloading the Wiimote stuff? The link you mention has an n instead of an m in it, which is stopping it find the right address: https://github.com/azzra/python3-wiimote

I'm going to close this comment for now as it is on a different repo, but please open a new issue on raspirobots if you are still having problems? Or feel free to directly contact me at: theraspberrypiguy@gmail.com

closed 2019-12-26 by the-raspberry-pi-guy