the-raspberry-pi-guy/lcd #9: Backlight control

HNJAMeindersma, 2020-05-26:
Is there any backlight control (e.g. on/off) available in your driver?

the-raspberry-pi-guy, 2020-07-06:
This hasn't been implemented in this repo, but if you add this support please feel free to make a PR and share! Thanks :)

closed 2020-07-06 by the-raspberry-pi-guy

HNJAMeindersma, 2020-09-16:
If I could, I would.
Possible feature request?

the-raspberry-pi-guy, 2020-09-18:
Sure, I understand. Unfortunately I do not have the time/resources to spend adding this feature for you - my studies are just about to recommence :)

HNJAMeindersma, 2020-10-09:
Understandable. Maybe I can splice some libraries together if I can find some.

Tomtom0201, 2020-10-16:
Check here for backlight control implementation (depending on your LCD hardware):
https://github.com/the-raspberry-pi-guy/lcd/issues/12#issuecomment-710183958