the-raspberry-pi-guy/lcd #11: turn off the backlight

jdarias, 2020-07-29:
Is it possible to turn off the backlight of the display?
I tried modifying your class adding these methods but it didn't work:
```python
   # backlight off
   def lcd_backlight_off(self):
       self.lcd_write(LCD_NOBACKLIGHT)

   # backlight on
   def lcd_backlight_on(self):
       self.lcd_write(LCD_BACKLIGHT)
```

the-raspberry-pi-guy, 2020-08-01:
I am not sure if this is available over the i2c interface, though there is support for it on the actual 16x2 boards I am pretty sure. It has been a while since I dipped my toes into this stuff, so I can't really advise any further. I did find this article if it is of any use: https://www.raspberrypi-spy.co.uk/2012/08/16x2-lcd-module-control-with-backlight-switch/

If you work out how to add this functionality then please submit a PR!

closed 2020-08-01 by the-raspberry-pi-guy

Tomtom0201, 2020-10-16:
Check here for backlight control implementation (depending on your LCD hardware):
https://github.com/the-raspberry-pi-guy/lcd/issues/12#issuecomment-710183958