the-raspberry-pi-guy/lcd #12: Backlight control

netrosec, 2020-10-13:
How about an example for controlling the backlight?

the-raspberry-pi-guy, 2020-10-13:
Please see #9 and #11

Tomtom0201, 2020-10-16:
**BACKLIGHT CONTROL** 😉✌️

--------------------
**STEP #1 - Adjustment**
edit **_lcddriver.py_**

- Add custom code to end of file
- Mind code indent

```
   # CUSTOM CODE
   # define backlight on/off (lcd_backlight(1); off= lcd_backlight(0)
   def lcd_backlight(self, state): # for state, 1 = on, 0 = off
      if state == 1:
         self.lcd_device.write_cmd(LCD_BACKLIGHT)
      elif state == 0:
         self.lcd_device.write_cmd(LCD_NOBACKLIGHT)
```

--------------------
**STEP #2 - Use backlight control in your code (.py)**
for example create new demo file **_demo_lcd_backlight.py_**

- Turn ON backlight >>> display.lcd_backlight(1)
- Turn OFF backlight >>> display.lcd_backlight(0)

```
# Simple string program. Writes and updates strings.
# Demo program for the I2C 16x2 Display from Ryanteck.uk
# Created by Matthew Timmons-Brown for The Raspberry Pi Guy YouTube channel
# Backlight: Enhanced by TOMDENKT - backlight control (on/off)
# Backlight: Caution - adjustments mandatory: add custom code to end of lcddriver.py file
# Backlight: Only copy custom code lines 8-14 into lcddriver.py (drop lines with quotation marks, mind code indent)
"""
   # CUSTOM CODE
   # define backlight on/off (lcd_backlight(1); off= lcd_backlight(0)
   def lcd_backlight(self, state): # for state, 1 = on, 0 = off
      if state == 1:
         self.lcd_device.write_cmd(LCD_BACKLIGHT)
      elif state == 0:
         self.lcd_device.write_cmd(LCD_NOBACKLIGHT)
"""

# Import necessary libraries for communication and display use
import lcddriver
import time

# Load the driver and set it to "display"
# If you use something from the driver library use the "display." prefix first
display = lcddriver.lcd()

# Main body of code
try:
    while True:
        # Remember that your sentences can only be 16 characters long!
        print("Writing to display and toggle backlight...")
        display.lcd_backlight(1)                          # Make sure backlight is on / turn on
        time.sleep(0.5)                                   # Waiting for backlight toggle
        display.lcd_backlight(0)                          # Turn backlight off
        time.sleep(0.5)                                   # Waiting for turning backlight on again
        display.lcd_backlight(1)                          # Turn backlight on again
        time.sleep(1)                                     # Waiting for text
        display.lcd_display_string("Greetings Human!", 1) # Write line of text to first line of display
        display.lcd_display_string("Demo Backlight", 2)   # Write line of text to second line of display
        time.sleep(2)                                     # Waiting for backlight toggle
        display.lcd_backlight(0)                          # Turn backlight off
        time.sleep(0.5)                                   # Waiting for turning backlight on again
        display.lcd_backlight(1)                          # Turn backlight on again
        time.sleep(0.5)                                   # Waiting for turning backlight off again
        display.lcd_backlight(0)                          # Turn backlight off
        time.sleep(0.5)                                   # Waiting for turning backlight on again
        display.lcd_backlight(1)                          # Turn backlight on again
        time.sleep(2)                                     # Give time for the message to be read
        display.lcd_display_string("I am a display!", 1)  # Refresh the first line of display with a different message
        time.sleep(2)                                     # Give time for the message to be read
        display.lcd_clear()                               # Clear the display of any data
        time.sleep(1)                                     # Give time for the message to be read
        display.lcd_backlight(0)                          # Turn backlight off
        time.sleep(1.5)                                   # Waiting until restart

except KeyboardInterrupt: # If there is a KeyboardInterrupt (when you press ctrl+c), exit the program and cleanup
    print("Cleaning up!")
    display.lcd_clear()
    display.lcd_backlight(1) # Make sure backlight is on / turn on by leaving
```

the-raspberry-pi-guy, 2020-10-16:
Hey @Tomtom0201 - nice stuff getting this to work (but only on some LCD hardware, is that right? What is the deal with this?)

If you've added functionality, the appropriate place to do this is in a pull request - that way I can approve your changes and they become part of the main repo too, rather than just a side note in a buried (closed) issue. Would you be able to rework this into some additional examples to include in the official repo and then submit a PR?

Tomtom0201, 2020-10-16:
Hi Matt,

I followed your YouTube instructions for adding a 16x2 LCD display to my Pi Zero and was searching for backlight control, too - so I came here. Due to the lack of any code examples I invented a solution by myself ;-)

**Regarding your questions:**
**1)** "but only on some LCD hardware, is that right?"
I used a kind of "standard" LCD 1602 display (HD44780) with I2C interface, followed your instructions and afterwards just **added lcd_backlight()**-function to **lcddriver.py**. Definitions for _LCD_BACKLIGHT_ resp. _LCD_NOBACKLIGHT_ state were already in place in **lcddriver.py**. So from this point: I think it will work for other displays with HD44780 chip/controller too, but I cannot completely confirm functionality under certain conditions. That's why this precaution _"may depend on your LCD hardware"_.

**2)** To be honest: I'm quite "new" to github development process flow ;-) But I will do my best to open a pull request to work amendments into it, so backlight control can become part of official repository.

Regards
Tom

the-raspberry-pi-guy, 2020-10-24:
Closing as functionality integrated by Tomtom0201

closed 2020-10-24 by the-raspberry-pi-guy