the-raspberry-pi-guy/lcd #17: Cant run install.sh

TiagoLourinho, 2020-12-23:
Hi!
I found your project on your tutorial (https://www.youtube.com/watch?v=fR5XhHYzUK0&t=281s&ab_channel=TheRaspberryPiGuy) but when i try to use it i get this error, and because i dont really know shell script i dont know what to do... I have  already tried to google the error but everything that i tried didnt work.
Thanks in advance, any help would be greatly appreciated!

[image]

cgomesu, 2020-12-23:
try `sudo bash install.sh` or `sudo ./install.sh` instead.  the line reference is a bash regex that `sh` won't understand.

TiagoLourinho, 2020-12-23:
> try `sudo bash install.sh` or `sudo ./install.sh` instead. the line reference is a bash regex that `sh` won't understand.

Thanks!! The first one worked! Closing the issue!

closed 2020-12-23 by TiagoLourinho

the-raspberry-pi-guy, 2020-12-24:
Good work again with helping this issue :))

PR #18 (2020-12-23): Hotfix for install shell compatibility issues (sh vs bash)
This fixes #17  as follows:
* The video tutorial and description tell users to run `install.sh` with `sh` instead of reading the interpreter from the shebang (`./install.sh`) or `bash` (`bash install.sh`).  This causes errors because the new version of the `install.sh` script has multiple *bashisms* (`[[`, `=~`, etc.).  Instead of changing the tutorial and descriptions, we can keep `install.sh` POSIX shell compliant and call the actual install script from there, using the correct interpreter (`bash`).  In this PR, this was achieved by  [... cut]