the-raspberry-pi-guy/lcd #21: ModuleNotFoundError: No module named 'i2c_dev'

AlcalaRegenera, 2020-12-29:
Hello, I used your code last year with Python2, now i wrote a new program in Python3 and i got this error, I have searched in the network but i don't found any result to fix this problem, sorry If i made any mistake writing in this part, i recently entered in github

I followed all your steeps from your youtube video.

Traceback (most recent call last):
  File "demo_clock.py", line 8, in <module>
    import drivers
  File "/home/pi/lcd/drivers/__init__.py", line 1, in <module>
    from i2c_dev import Lcd
ModuleNotFoundError: No module named 'i2c_dev'

Thanks!

cgomesu, 2020-12-29:
Just saw your message as I'm working on a script right now as well.  The error is a relative vs, absolute path issue related to py3.  In `drivers/__init__.py`, just change `i2c_dev` for `.i2c_dev` and you should be good to go.  (I think we can add such a change  to the repo as well because it won't affect Python2 execution. I'll send a PR tomorrow if no one is able to do so before then.)  Also, remember that you'll need to install `python3-smbus` if running Python 3.  Otherwise, you'll see a message complaining about the missing `smbus` pkg.  The install script only installs `smbus` for py2.

AlcalaRegenera, 2020-12-29:
hello, I followed your steeps, it is working now, i am very happy and i really thanks you :) 

good work :)

Solved.

piderking, 2020-12-30:
I was having a little trouble with this at first, but It works now. Thanks, good work.

PR #23 (2020-12-31): Hotfix for i2c_dev import issue when using Python3+
This fixes #21 as follows:
* It adds a relative path to `i2c_dev` in the `__init__.py` file that allows to initialize `Lcd` when running Python 3+.  This change does not affect Python 2.7.

As a side note, `setup.sh` only installs the `smbus` package for Python2.  Users trying to run code with Python 3+ will have to manually install `python3-smbus`.  To avoid this issue, you can simply add the latter pkg to the `apt install` list in `setup.sh`.  (Didn't want to do that here because it's not directly related to issue #21.)
closed 2020-12-31 by the-raspberry-pi-guy