the-raspberry-pi-guy/lcd #24: Parameter Explanation

viridianajim, 2021-05-12:
Hi

I am working on a small project, and  I need to display an ID and a name into my LCD screen 

I am using **display.lcd_display_string("TEXT", 1)**

I have been testing different numbers; however, the second **display.lcd_display_string("TEXT2", 1)** that I use does not display the text as I want. 

Do you have any documentation that explains the parameters of this method?

Thanks

cgomesu, 2021-05-12:
> 

``
``

cgomesu, 2021-05-12:
> Do you have any documentation that explains the parameters of this method?

I don't think so, @viridianajim .  It's pretty straightforward though.  You're probably not cleaning the LCD before the second write.   Check the comments in the `i2c_dev.py` file, specifically:

```
    # put string function
    def lcd_display_string(self, string, line):
        if line == 1:
            self.lcd_write(0x80)
        if line == 2:
            self.lcd_write(0xC0)
        if line == 3:
            self.lcd_write(0x94)
        if line == 4:
            self.lcd_write(0xD4)
        for char in string:
            self.lcd_write(ord(char), Rs)

    # clear lcd and set to home
    def lcd_clear(self):
        self.lcd_write(LCD_CLEARDISPLAY)
        self.lcd_write(LCD_RETURNHOME)
```

Check the code in the demos as well.  Their implementation should give you an idea how to display the IDs for your project.

the-raspberry-pi-guy, 2021-07-14:
Thanks for the reply to this @cgomesu :) Closing the issue

closed 2021-07-14 by the-raspberry-pi-guy