the-raspberry-pi-guy/lcd #26: initial install Errno 121

rnld-rygn, 2021-07-31:
Thanks for this great project. I am in the process of running this installation on a pi 3b+ running the latest raspbian 32bit. Linux raspberrypi 5.10.17-v7+ #1414

I have used your README for setup and everything seems to install correctly. At first run of demo_clock.py though I receive an Errno121 the full error is below.

Traceback (most recent call last):
  File "demo_clock.py", line 14, in <module>
    display = drivers.Lcd()
  File "/home/pi/lcd/drivers/i2c_dev.py", line 103, in __init__
    self.lcd_write(0x03)
  File "/home/pi/lcd/drivers/i2c_dev.py", line 126, in lcd_write
    self.lcd_write_four_bits(mode | (cmd & 0xF0))
  File "/home/pi/lcd/drivers/i2c_dev.py", line 121, in lcd_write_four_bits
    self.lcd.write_cmd(data | LCD_BACKLIGHT)
  File "/home/pi/lcd/drivers/i2c_dev.py", line 74, in write_cmd
    self.bus.write_byte(self.addr, cmd)
IOError: [Errno 121] Remote I/O error

If I run i2cdetect -y 1 I get an address of 3f, so I know the hardware is at least being seen. I used a simpler guide/setup and was able to edit the text the lcd displays with this https://www.circuitbasics.com/raspberry-pi-i2c-lcd-set-up-and-programming/ I did this on a second microsd as to not interfere with the pi-guy version. So I know the hw is functioning and wired correctly. Anyone have any suggestions please?

Thank you!

rnld-rygn, 2021-07-31:
I think I might have some sort of hw problem. I have successfully got it to run once or twice but it eventually fails. I ended up changing line 102 in the lcd/drivers/i2c_dev.py file to the address i2cdetect -y 1 would give
self.lcd = I2CDevice(addr_default=0x3f) 
This will work for one command, I then try to run it a second time and receive the same error! Here is where it gets weird though. If I run i2cdetect -y 1 again, I get a different address, this time it is 3b. It seems to fluctuate between the two. If I edit i2c_dev.py with the 3b address the lcd will show the no time to waste but it fails immediately with the errno 121 again. I then do another i2cdetect -y 1 and it shows back to the 3f address. If I edit i2c_dev.py to the correct address, yet again it will work again once but then no longer works after the CTRL+C and the address flips again to 3b.

Does that sound like hardware failure?

cgomesu, 2021-08-01:
> If I run i2cdetect -y 1 I get an address of 3f, so I know the hardware is at least being seen. 

please post the output of the `i2cdetect -y 1` command.

> I think I might have some sort of hw problem. I have successfully got it to run once or twice but it eventually fails. I ended up changing line 102 in the lcd/drivers/i2c_dev.py file to the address i2cdetect -y 1 would give
self.lcd = I2CDevice(addr_default=0x3f)

you don't need to edit the driver file. just set the `addr` attribute when loading the `Lcd()` class in your application. in https://github.com/the-raspberry-pi-guy/lcd/blob/master/demo_clock.py#L14, for example, use the following instead to set the Lcd address to `0x3f` (mentioned in #20):

```python
display = drivers.Lcd(addr=0x3f)
```

either way, code was supposed to autodetect the address. waiting for your reply to the first question to troubleshoot this further.

rnld-rygn, 2021-08-02:
Thanks so much for the assistance. 

I just wiped and reloaded raspberry pi os 32-bit headless so I can start from scratch. I followed the instructions to the letter, the only deviation was an apt update.

`i2cdetect -y 1`
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3f 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

If I try to run demo_clock.py from the lcd directory I get this.

`python demo_clock.py`
Traceback (most recent call last):
  File "demo_clock.py", line 14, in <module>
    display = drivers.Lcd()
  File "/home/pi/lcd/drivers/i2c_dev.py", line 103, in __init__
    self.lcd_write(0x03)
  File "/home/pi/lcd/drivers/i2c_dev.py", line 126, in lcd_write
    self.lcd_write_four_bits(mode | (cmd & 0xF0))
  File "/home/pi/lcd/drivers/i2c_dev.py", line 121, in lcd_write_four_bits
    self.lcd.write_cmd(data | LCD_BACKLIGHT)
  File "/home/pi/lcd/drivers/i2c_dev.py", line 74, in write_cmd
    self.bus.write_byte(self.addr, cmd)
IOError: [Errno 121] Remote I/O error

I found that if I changed line 102 in /home/pi/lcd/drivers/i2c_dev.py to the 3f it fixed it. I know this isn't the correct way to run it. But it at least confirms the display is functioning correctly.

cgomesu, 2021-08-03:
okay. I'm investigating why it is not auto-detecting your `addr`. will update once I find out.

cgomesu, 2021-08-03:
found the culprit at https://github.com/the-raspberry-pi-guy/lcd/blob/master/drivers/i2c_dev.py#L60-L70:
```python
        if not addr:
            # try autodetect address, else use default if provided
            try:
                self.addr = int('0x{}'.format(
                    findall("[0-9a-z]{2}(?!:)", check_output(['/usr/sbin/i2cdetect', '-y', BUS_NUMBER]))[0]), base=16) \
                    if exists('/usr/sbin/i2cdetect') else addr_default
            except:
                self.addr = addr_default
        else:
            self.addr = addr
        self.bus = SMBus(bus)
```
in `check_output()`, args must be strings but `BUS_NUMBER` is an integer. this causes an error and sets `self.addr = addr_default`, which is set to the hex literal `0x27`.  in @rnld-rygn 's case, this will cause errors downstream because their i2c device has a different address, namely `0x37`.

setting `str(BUS_NUMBER)` in https://github.com/the-raspberry-pi-guy/lcd/blob/master/drivers/i2c_dev.py#L64 should fix this.  will double check it later this week and submit a PR.

in the meantime, as I mentioned before, you can avoid this issue altogether by setting `Lcd(addr=0x3f)` in your application.

rnld-rygn, 2021-08-04:
Thanks so much for your help. I will try the recommended fix you gave and keep an eye out for the release fix.

closed 2021-08-04 by rnld-rygn

the-raspberry-pi-guy, 2021-08-05:
Top notch stuff @cgomesu, thanks for supporting this repo :)