the-raspberry-pi-guy/lcd #41: smbus is not recognized

AndyBaBaBoi, 2022-01-04:
**Describe the bug**
When I type ‘python demo_lcd.py’ there is an error where it says that in line 1 of __init__ or somewhere in drivers that ‘from smbus import SMBus’ smbus is not reconized or smth.

**Executed command and associated error**
```sudo python3 demo_lcd```
add the executed command here
```python demo_lcd
```
add the associated error here
```Type error: smbus not reconized```

**Host and software info**
- RPi board version: `latest one`
- OS version: `raspian latest one`
- (If Python related.) Python version : `version 3`
- (If shell related.) Shell: `add name here and version`

**Checklist**
- [ ] I have watched and followed the instructions in the [Youtube tutorial](https://www.youtube.com/watch?v=fR5XhHYzUK0).
- [ ] I have searched [open and closed issues](https://github.com/the-raspberry-pi-guy/lcd/issues) for either an identical or similar bug before reporting this new one.
- [ ] (If Python related.) I have used Python 2.7

**Additional context**
Add any other context about the problem here.

cgomesu, 2022-01-04:
> ```
> Type error: smbus not reconized
> ```

paste the entire output instead

> Python version : `version 3`

be more specific. `3.5`, `3.7`, `3.9`, `3.10`?

> - [ ] (If Python related.) I have used Python 2.7

here is the culprit.  the lcd drive and demos were meant to be used with Python 2.7.  Python 3.x may or may not work.  in any case, the error you are seeing has to do with a missing `smbus` pkg in your Python 3.x standard library.  you can try installing it manually to fix the issue:

```
sudo apt update && sudo apt install python3-smbus
```

AndyBaBaBoi, 2022-01-04:
Ok thx ill try that after school…and tell the creator of this that he/she is a legend for making this.

cgomesu, 2022-01-04:
> Ok thx ill try that after school…and tell the creator of this that he/she is a legend for making this.

sounds good, @AndyBaBaBoi .  the repo was created by Matt (@the-raspberry-pi-guy).  I just help maintaining it.

> Oh also I just looked through some of the closed issues and there were a few people with the same issue

if you are referring to #42, #43, and #44, their issue seems to be different because they were using `python 2.7.18` instead of `python 3.x`, like yourself. 

> so it may be wise to say that if u r on x versions then do y if this error shows up or smth

changes that improve compatibility are always welcomed.  to contribute to the project, please read our [CONTRIBUTING.md](https://github.com/the-raspberry-pi-guy/lcd/blob/master/CONTRIBUTING.md) guide.

AndyBaBaBoi, 2022-01-04:
ok so @cgomesu i went ahead and did as you recommended and this happened:

`pi@raspberrypi:~/lcd $ sudo apt update && sudo apt install python3-smbus
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-smbus is already the newest version (4.2-1+b1).
The following packages were automatically installed and are no longer required:
  libfuse2 python-is-python2
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

so i then went ahead and did this:

`pi@raspberrypi:~/lcd $ python demo_lcd.py
Traceback (most recent call last):
  File "demo_lcd.py", line 8, in <module>
    import drivers
  File "/home/pi/lcd/drivers/__init__.py", line 1, in <module>
    from .i2c_dev import Lcd, CustomCharacters
  File "/home/pi/lcd/drivers/i2c_dev.py", line 1, in <module>
    from smbus import SMBus
ImportError: No module named smbus`

do you know why it is not connecting reconizing smbus?

cgomesu, 2022-01-04:
paste the output of the following:

```
hostnamectl
```
```
python --version
```
```
pip list
```
```
python3 --version
```
```
pip3 list
```

AndyBaBaBoi, 2022-01-04:
ok

hostnamectl:
'   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: ******************************************
           Boot ID: ******************************************
  Operating System: Raspbian GNU/Linux 11 (bullseye)
            Kernel: Linux 5.10.63-v7l+
      Architecture: arm`

python --version:
`Python 2.7.18`

pip list:
`Package           Version
----------------- ---------
appdirs           1.4.4
arandr            0.1.10
astroid           2.5.1
asttokens         2.0.4
automationhat     0.2.0
backcall          0.2.0
beautifulsoup4    4.9.3
blinker           1.4
blinkt            0.1.2
buttonshim        0.0.2
Cap1xxx           0.1.3
certifi           2020.6.20
chardet           4.0.0
click             7.1.2
colorama          0.4.4
colorzero         1.1
cryptography      3.3.2
cupshelpers       1.0
cycler            0.10.0
dbus-python       1.2.16
decorator         4.4.2
distro            1.5.0
docutils          0.16
drumhat           0.1.0
envirophat        1.0.0
ExplorerHAT       0.4.2
Flask             1.1.2
fourletterphat    0.1.0
gpiozero          1.6.2
guizero           1.1.1
gyp               0.1
html5lib          1.1
idna              2.10
ipykernel         5.4.3
ipython           7.20.0
ipython-genutils  0.2.0
isort             5.6.4
itsdangerous      1.1.0
jedi              0.18.0
Jinja2            2.11.3
jupyter-client    6.1.11
jupyter-core      4.7.1
kiwisolver        1.3.1
lazy-object-proxy 0.0.0
logilab-common    1.8.1
lxml              4.6.3
MarkupSafe        1.1.1
matplotlib        3.3.4
mccabe            0.6.1
mfrc522           0.0.7
microdotphat      0.2.1
mock              4.0.3
mote              0.0.4
[... 199 lines cut ...]
sense-hat         2.2.0
setuptools        52.0.0
simplejson        3.17.2
six               1.16.0
skywriter         0.0.7
sn3218            1.2.7
soupsieve         2.2.1
spidev            3.5
ssh-import-id     5.10
thonny            3.3.14
toml              0.10.1
tornado           6.1
touchphat         0.0.1
traitlets         5.0.5
twython           3.8.2
typed-ast         1.4.2
typing-extensions 3.7.4.3
uflash            1.2.4
unicornhathd      0.0.4
urllib3           1.26.5
wcwidth           0.1.9
webencodings      0.5.1
Werkzeug          1.0.1
wheel             0.34.2
wrapt             1.12.1

cgomesu, 2022-01-04:
okay.  will take a closer look at this issue either tomorrow or Thursday.
it's not compatibility though.

cgomesu, 2022-01-05:
found the issue (also related to #42, #43, #44):

- `python-smbus` is missing from the raspberry pi `main` branch in **`bullseye`** but not `buster` or previous dists.  (see https://archive.raspbian.org/raspbian/dists/bullseye/main/binary-armhf/.)

working on a solution via a change to [`setup.sh`](https://github.com/the-raspberry-pi-guy/lcd/blob/master/setup.sh).

---

in the meantime, @AndyBaBaBoi (and @JoshyOnMars), if you have already installed `pyhton3-smbus`, you should be able to use the lcd driver with `python3` instead.  for example, to run the demo `demo_lcd.py`, use the following:

```
python3 demo_lcd.py
```

please **do not close this issue** if the above solution works for you.  I'll close it once I merge a fix for the `python2` issue.

AndyBaBaBoi, 2022-01-05:
@cgomesu I just went ahead and tried it like you said to and I got this error. is this because of a compatability issue or smth?

pi@raspberrypi:~ $ cd lcd
pi@raspberrypi:~/lcd $ python3 demo_lcd.py
Traceback (most recent call last):
  File "/home/pi/lcd/demo_lcd.py", line 13, in <module>
    display = drivers.Lcd()
  File "/home/pi/lcd/drivers/i2c_dev.py", line 104, in __init__
    self.lcd_write(0x03)
  File "/home/pi/lcd/drivers/i2c_dev.py", line 127, in lcd_write
    self.lcd_write_four_bits(mode | (cmd & 0xF0))
  File "/home/pi/lcd/drivers/i2c_dev.py", line 122, in lcd_write_four_bits
    self.lcd.write_cmd(data | LCD_BACKLIGHT)
  File "/home/pi/lcd/drivers/i2c_dev.py", line 74, in write_cmd
    self.bus.write_byte(self.addr, cmd)
OSError: [Errno 121] Remote I/O error
pi@raspberrypi:~/lcd $

AndyBaBaBoi, 2022-01-05:
@cgomesu ok so it works in printing that it is writing to display but it is not showing up on the screen

AndyBaBaBoi, 2022-01-05:
https://user-images.githubusercontent.com/97101062/148291737-173b93b9-31a4-43f3-96df-13299e4e1452.MOV

AndyBaBaBoi, 2022-01-06:
OMG I GOT IT TO WORK i literally twisted the thing on the back the little plus sign BUT IT WORKS

cgomesu, 2022-01-06:
> YESSSSS I GOT IT TO WORK i literally twisted the thing on the back the little plus sign BUT IT WORKS

this was explained in the tutorial: https://www.youtube.com/watch?v=fR5XhHYzUK0&t=410s

AndyBaBaBoi, 2022-01-06:
Oh lol i literally watched until literally 1 second before it but also do u think the display could be able to rick roll someone?

cgomesu, 2022-01-06:
> Oh lol i literally watched until literally 1 second before it but also do u think the display could be able to rick roll someone? I mean it would take a bit of work for the custom characters for that but it would be worth it

there's a lot that can be done but you will be working with a *very* limited amount of memory. per @juvus' comments in the [demo_lcd_custom_characters.py](https://github.com/the-raspberry-pi-guy/lcd/blob/master/demo_lcd_custom_characters.py) example,  you can define and store custom characters in the custom generated random access memory (cg-ram) of the lcd controller, which is 64 Bytes long and because each character uses 8 Bytes, you can only define 8 custom characters that way.

AndyBaBaBoi, 2022-01-06:
Ohhhh I see. So would that in theory mean that each frame of the “rick roll” would have to use less than 8 custom characters per frame? Like i could make an infinite amount of them but only load up to eight at a time or can i only create 8 overall. If that is the case could I do some thing where like each frame x list becomes cc1 and y list becomes cc2 and i can replace the 8 with new ones each frame or is it just not possible

juvus, 2022-01-07:
> Ohhhh I see. So would that in theory mean that each frame of the “rick roll” would have to use less than 8 custom characters per frame? Like i could make an infinite amount of them but only load up to eight at a time or can i only create 8 overall. If that is the case could I do some thing where like each frame x list becomes cc1 and y list becomes cc2 and i can replace the 8 with new ones each frame and with that I would have to load it to cg each time?

Hi, AndyBaBaBoi.
Of course you can define any amount of you own characters and store it, for example, in the python list or dictionary, but you can upload to the LCD cg-ram no more than 8 characters data at a time. During the work of your program you can easily upload new caracters data to the cg-ram (by invoking **load_custom_characters_data()** method). But, please, be noted, that if you redefine any custom character in the memory it will be immideatly updated on the LCD screen with the new character data. So, the general conclusion here is that _it is not possible to display more that 8 custom characters on the display at the same frame (at the same time)._

AndyBaBaBoi, 2022-01-07:
OK I see now but just to clarify:

This will not work as when I do ```cc.load_custom_characters_data()``` it would try to load 9 cc and not 8 as it is supposed to?
```
cc.char_8_data = ["11111",
                              "10001",
                              "11111",
                              "00000",
                              "00000",
                              "11111",
                              "10001",
                              "11111"]
```
But I am able to do this and to then do ```cc.load_custom_characters_data()```
```
custom_char_8 = ["11111",
                              "10001",
                              "11111",
                              "00000",
                              "00000",
                              "11111",
                              "10001",
                              "11111"]
cc.char_8_data = custom_char_8
```

AndyBaBaBoi, 2022-01-10:
Also @cgomesu how do i use it with these pinns? 
[image]

cgomesu, 2022-01-10:
@AndyBaBaBoi, please keep the discussion in this thread about the reported bug.  if you have unrelated questions, open a new issue.  be mindful that most of us do this in their spare time, so we won't be able to help on a regular basis.

juvus, 2022-01-10:
Hi, @AndyBaBaBoi!
As I understood from the presious conversations, you are a schollboy - young person which is interested in electronics, programming and with a lot of questions about different issues. This is very good, and a young person should be like this! Try to find the answers to you questions by you own, try to figure out how to make a good questions to Google, find information, weed out unrelated info and so on. This is very important skill nowadays. Try to be independent! Only this way you will became a very quilified professional in whatever field you want! I wish you good luck!

closed 2022-01-14 by cgomesu