the-raspberry-pi-guy/lcd #51: turn off the lcd screen

bob5731, 2022-05-30:
How do I turn off a lcd screen?

cgomesu, 2022-05-30:
take a look at the backlight control demo

bob5731, 2022-05-30:
Can I get some example code?

cgomesu, 2022-05-30:
```Python
import drivers
from time import sleep

display = drivers.Lcd()

#turn the backlight on
display.lcd_backlight(1)
#wait for the backlight to toggle
sleep(0.5)
#turn the backlight off
display.lcd_backlight(0)
```

from [demo_lcd_backlight.py](https://github.com/the-raspberry-pi-guy/lcd/blob/master/demo_lcd_backlight.py)

bob5731, 2022-05-31:
Package 'python-smbus' has no installation candidate
[LCD] [WARNING] Unable to install pkg 'python-smbus'.

bob5731, 2022-05-31:
> ```python
> import drivers
> from time import sleep
>
> display = drivers.Lcd()
>
> #turn the backlight on
> display.lcd_backlight(1)
> #wait for the backlight to toggle
> sleep(0.5)
> #turn the backlight off [...]

File "/home/pi/pill-dispenser/test12.py", line 1, in <module>
    import drivers
ModuleNotFoundError: No module n

closed 2022-06-07 by cgomesu