the-raspberry-pi-guy/lcd #55: module not imported running python script at startup raspberry pi  /etc/profiles

Murali-IsPagro, 2022-07-27:
I created a python script using lcd. It works perfectly fine when I am testing from CLI. I have an application where I have to run the script during start up. So I have done these steps:

1. sudo nano /etc/profile

2. added "sudo python3 /home/pi/test_adc_1.py &" in the last line.

3. sudo reboot now

But code didnt start, so i logged on to RPi and checked it displays below error.

**pi@raspberrypi:~ $ Traceback (most recent call last):
  File "/home/pi/test_adc_1.py", line 3, in <module>
    import drivers
ModuleNotFoundError: No module named 'drivers'

[1]+  Exit 1                  sudo python3 /home/pi/test_adc_1.py**

Kindly suggest me the solution.

cgomesu, 2022-07-28:
> sudo nano /etc/profile

try https://github.com/the-raspberry-pi-guy/lcd#systemd instead.  a systemd service unit file is a better way of running such things.

> ```python
> Traceback (most recent call last):
> File "/home/pi/test_adc_1.py", line 3, in import drivers
> ModuleNotFoundError: No module named 'drivers'
> ```

see https://duckduckgo.com/?t=ffab&q=python+import+relative+absolute+paths.

Murali-IsPagro, 2022-07-29:
This one fixed the error. Now working fine. 

Thanks!

closed 2022-07-29 by cgomesu