the-raspberry-pi-guy/lcd #56: Cannot run install.sh

vismodo, 2022-12-31:
**Describe the bug**
Cannot run install.sh

**Executed command and associated error**
```
sudo sh install.sh # i also tried using sudo install.sh
```
```
install.sh: 21: ./setup.sh: Permission denied
```

**Host and software info**
- RPi board version: `Pi 4B`
- OS version: `Raspberry PI OS 64 bit released 2022-09-22`
- (If shell related.) Shell: `GNU bash, version 5.1.4(1)-release (aarch64-unknown-linux-gnu)`

**Checklist**
- [x] I have watched and followed the instructions in the [Youtube tutorial](https://www.youtube.com/watch?v=fR5XhHYzUK0).
- [x] I have searched [open and closed issues](https://github.com/the-raspberry-pi-guy/lcd/issues) for either an identical or similar bug before reporting this new one.

**Additional context**
I have not used the raspberry pi without ssh. The pi has never been setup. I use a Pi 4B and the official 64 bit rpi os.

warrence, 2022-12-31:
im having the same problem too

cgomesu, 2022-12-31:
Do you still get permission denied if you try running `setup.sh` with sudo?

```sh
sudo bash setup.sh
```

The only thing that the `install.sh` script does is call `setup.sh`, which actually installs all related packages and enables I2C.

bvirg, 2023-01-02:
chmod u+x setup.sh

cgomesu, 2023-01-03:
@vismodo, @warrence: please confirm that @bvirg's suggestion fixes the issue.

cgomesu, 2023-01-06:
Please update your local repo (`git pull`) and try running the `install.sh` once again.  If the issue persists, let me know.  I will assume it has been fixed for the time being.

PR #57 (2023-01-06): Fix default execute permission
- Fix #56 

Add execute permission to remaining files, including `setup.sh`.
closed 2023-01-06 by cgomesu