the-raspberry-pi-guy/lcd #58: would like the cpu/gpu temp display code

SgtKiLLx, 2023-02-04:
hello . im wondering if you have the code to display the pi's cpu/gpu temp on the 1602 i2c

cgomesu, 2023-02-04:
this is not an issue related to the content of this repo.  however, I can suggest you to read about the [sysfs](https://www.kernel.org/doc/html/latest/filesystems/sysfs.html).  there's a subsystem called `thermal` in `/sys/class/` that should contain one or more `thermal_zone` for different thermal sensors (e.g., `thermal_zone0`). each `thermal_zone` should have a `temp` file that stores the current temperature in millidegree Celsius.  so, all you need to do is to inspect such a file with your Python script.

closed 2023-02-04 by cgomesu

SgtKiLLx, 2023-02-04:
wondering if you could help a borderline noob with this

cgomesu, 2023-02-05:
@SgtKiLLx , you can implement this in many different ways.  take a look at the demos for examples of how to interface with the LED driver.

regarding my previous comment, you can use the `open()` method in Python to assign to a variable the current temperature of a thermal zone, like this:

```python
# create a variable to store the current temperature of the zone0 thermal sensor
temp_zone0 = None

# get the current value from sysfs and assign to your temp_zone0 variable
with open("/sys/class/thermal/thermal_zone0/temp") as temp:
    temp_zone0 = float(temp.read())/1000

# now you can do whatever you want with temp_zone0
if temp_zone0:
   print(f"The current temperature of the thermal_zone0 sensor is {temp_zone0} degrees Celsius")
else:
   print("Unable to read the temperature of the thermal_zone0 sensor")
```

however, instead of `print`, you have to insert a logic to display the string on the LED. there are many examples in the demos that illustrate how to do exactly that. in addition, you might want to wrap such code in a loop (`for`, `while`) to display new temperature readings over time. for a better implementation, you would wrap the `with` statement in a try block to catch an `OSError` exception that `open()` might raise.