the-raspberry-pi-guy/lcd #59: Display IP adress

Mikes-Git, 2023-03-04:
Displaying the IP Address shows following:

b'[IP Address]'

how can I get the b and the '' no to be displayed?

cgomesu, 2023-03-04:
user ignored the issue template.

closed 2023-03-04 by cgomesu

Mikes-Git, 2023-03-04:
Help out a noop, what did I do wrong?

Sierra007117, 2023-03-06:
The 'b' prefix in front of the IP address indicates that it is a bytes object, rather than a string. To remove the prefix and the quotes, you can use the decode() method to convert the bytes object to a string. 

As for the port, between the protocols User Datagram Protocol (UDP) and Transmission Control Protocol (TCP), there are 65,535 ports available for communication between devices. If you meant the port used for SSH, you could modify the check_output command to run the sshd command with the -T flag, which displays the port number used for SSH.