the-raspberry-pi-guy/lcd #68: LCD Spotify Now Playing + Telegram Bot

Jumitti, 2023-12-19:
Hello everyone

First of all, I apologize if this is not the right place to post this.

I didn't find any bugs or anything but I coded a "small" script. So I would like to know if you are interested in having it in demo or not. Besides, I think that if it interests you, you will perhaps ask me to separate it into several small demos.

Let me explain. I manage Telegram bots quite well. The advantage is that you can run these bots on a Raspberry and communicate with them to execute commands on the Raspberry (even when you are not at home and you cannot make SSH connections). I use it to know the temperature of my CPU when the Raspberry is doing complicated calculations or something else, to make it do updates and also different little things. In fact, as long as you code your bot well, you can make it do what you want on Raspberry since it's python. I then decided to add a DS18B20 thermal probe to my Raspberry (because the CPU heats up okay but in my house it's cold so I might as well know if I'm going to die of hypothermia soon or not). In addition to that, I added additional fans if the CPU temperature rises too much to cool it down.

Well and with all that, I found it stupid not to have a nice little display. So, I bought a 16x2 LCD and I made my first script for the LCD:
- Display the date, time, CPU temperature and my house temperature.

At that time I was starting out in coding (this dates back to 2019) so the Telegram bot and the LCD script were not in one and the same script and that caused a lot of problems. I have since fixed it.

I wanted to go further so I officially implemented the Telegram bot in the LCD script and it allows you to turn the backlight on and off as you wish only via Telegram. But theoretically, you can write personalized messages (for flirting it's perfect) etc etc..

And finally, when I code I have music playing loudly. What if we added a function that allows us to display what we are listening to on Spotify? It's done too

Overall, I know that there are a lot of functions and that it goes in all directions. And that's why I prefer to see if it might interest people or not :)

Let me know what you think about it :)

Julien

cgomesu, 2023-12-20:
new demos are always welcomed, @Jumitti . please read our [contributing guidelines](https://github.com/the-raspberry-pi-guy/lcd/blob/f6e719c12187da33396b125bff2fae34ee36d487/CONTRIBUTING.md) first.  you can use the other demos as a template but feel free to go beyond if your implementation requires slightly more complex code.  when you're ready, just submit a PR to the main branch (add `-fix #68` to the description to link it to this issue) and this will start the formal review process.

Jumitti, 2023-12-23:
Hello @cgomesu 

So I use API for my script. So it requires tokens and IDs for connections. I usually use a SECRETS.json to store this information and not have it written directly in the script. It’s security.

But over time, I got into the habit of putting all the script variables in there. This allows certain parameters to be modified quickly without stopping my scripts.

Do you think I should do the same for the LCD? I don't know if it will be user friendly enough

cgomesu, 2023-12-23:
that could work but notice that all demos are a single demo_*.py file. if
your implementation depends on external files, they won't be merged into
the master branch of this repo. this also means you'll have to spend more
time documenting the demo (making the necessary changes to readme) to
explain how to properly use it.

on a second thought, I think you should consider creating your own repo for
your implementation. once it's done, we can add a reference to it in the
documentation. that would as well.

CG

Jumitti, 2023-12-24:
Hello @cgomesu 

Sorry to bother you again during the holiday season.

I think I'm done with my scripts and documentation. If you have a moment or the desire, take a look at my Github repo https://github.com/Jumitti/lcd_home_automation
I actually think that the simplest thing would be to just put a reference in the documentation of the official repo, as you suggested yes.
If you find anything missing, let me know.

Happy Christmas and Happy New Year

JM

cgomesu, 2023-12-25:
that looks good. loved the in-depth documentation. if I may suggest you a couple of things: 

I think it would look better your repo focused just on your implementation. that is, remove all the other demos and unrelated files (`.github`, `CONTRIBUTING.md`, edit the About, etc) and  remove the `demo_` prefix from your scripts because I think it's more than just a demo. 

once you are done with all that, ping me here and I'll edit the `README.md` to reference your implementation. (happy holidays to you too!)

Jumitti, 2023-12-26:
Hello @cgomesu 

I think I'm done with the documentation etc of Home Automation. If you see anything else, don't hesitate to tell me.

I also created a small Streamlit application (web deployment of a python script) to facilitate the creation of custom characters. I didn't do a fork for that, here is the Github https://github.com/Jumitti/lcd_custom_characters and the Streamlit application https://custom-characters-for-lcd16x2.streamlit.app/
I think maybe this might interest people, maybe

Best,

Julien

cgomesu, 2023-12-27:
awesome! I`ll push a change in the documentation to reference your implementations asap.

PR #72 (2023-12-27): Home automation documentation + markdown linting fix
- fix #68:
  - home automation demo
  - custom chars web app
- also took the opportunity to fix a few md formatting issues to improve readability
closed 2023-12-27 by cgomesu