the-raspberry-pi-guy/lcd #73: cc.load_custom_characters_data() not working

Jimmy106-e, 2024-03-17:
**Describe the bug**
Im trying the custom character functionality with a (not really) bitmap and also just writing it and it is giving me an error.

**Executed command and associated error**
```
cc.load_custom_characters_data()
```
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/panda/lcd/Bitmap editor test/Bitmap editor test/web-server.py", line 17, in get_bitmap
    cc.load_custom_characters_data()
  File "/home/panda/lcd/Bitmap editor test/Bitmap editor test/drivers/i2c_dev.py", line 284, in load_custom_characters_data
    self.lcd.lcd_write(int(binary_str_cmd, 2), Rs)
                       ^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 2: '0b000"00000", "00000", "00000", "00100", "00000", "00000", "00000", "00000"'
```

**Host and software info**
- RPi board version: `revision 1.2`
- OS version: 12`
- (If Python related.) Python version : `3.11.2`

**Checklist**
- [ /] I have watched and followed the instructions in the [Youtube tutorial](https://www.youtube.com/watch?v=fR5XhHYzUK0).
- [ ] I have searched [open and closed issues](https://github.com/the-raspberry-pi-guy/lcd/issues) for either an identical or similar bug before reporting this new one.
- [ ] (If Python related.) I have used Python 2.7

**Additional context**
It is a Flask web server program with a (lets say) bitmap editor to write custom character data.

cgomesu, 2024-03-18:
@juvus could you please take a look at this? it's related to your custom character implementation. (@Jumitti might be able to help as well. Feel free to chime in.)

Jumitti, 2024-03-18:
@Jimmy106-e can you show me your code please? Also, I refer you to my GitHub repository, you may find the solution there https://github.com/Jumitti/lcd_home_automation

Jimmy106-e, 2024-03-19:
> @Jimmy106-e can you show me your code please? Also, I refer you to my GitHub repository, you may find the solution there https://github.com/Jumitti/lcd_home_automation

Python code (With Flask)
```
from flask import Flask, render_template, request
import drivers

app = Flask(__name__)
display = drivers.Lcd()
cc = drivers.CustomCharacters(display)

@app.route('/')
def index():
    return render_template('index.html')

@app.route('/get_bitmap', methods=['POST'])
def get_bitmap():
    bitmap_data = request.form['bitmapData']
    print("Bitmap data:", bitmap_data)
    cc.char_1_data = [bitmap_data]
    #cc.load_custom_characters_data()
    print(cc.char_1_data)
    #display.lcd_display_extended_string("{0x00}",1)
    return '', 204

if __name__ == '__main__':
    app.run('0.0.0.0')

```

HTML (in folder called "templates")
```
<!DOCTYPE html>
<html lang="en">
<head>
  <meta charset="UTF-8">
  <meta name="viewport" content="width=device-width, initial-scale=1.0">
  <title>Bitmap Editor</title>
  <style>
    .grid {
      display: grid;
      grid-template-columns: repeat(5, 30px);
      grid-template-rows: repeat(8, 30px);
      gap: 1px;
    }
    .cell {
      width: 30px;
      height: 30px;
      background-color: white;
      border: 1px solid black;
    }
    .cell.active {
      background-color: black;
    }
  </style>
</head>
<body>
  <form action="/get_bitmap" method="post">
    <h1>Bitmap Editor</h1>
    <div class="grid">
      {% for row in range(8) %}
        {% for col in range(5) %}
          <div class="cell" onclick="toggleCell(this)"></div>
        {% endfor %}
      {% endfor %}
    </div>
    <button type="submit">Get Bitmap</button>
    <input type="hidden" name="bitmapData" id="bitmapData">
  </form>

  <script>
    function toggleCell(cell) {
      cell.classList.toggle('active');
    }

    document.querySelector('form').addEventListener('submit', function(event) {
      event.preventDefault();
      const bitmapData = Array.from(document.querySelectorAll('.cell')).map(cell => cell.classList.contains('active') ? 1 : 0).join('');
      
      // Insert comma and quotation marks
      let bitmapString = '' ;
      for (let i = 0; i < bitmapData.length; i++) {
        if (i % 5 === 0) bitmapString += '"';
        bitmapString += bitmapData[i];
        if ((i + 1) % 5 === 0 && i !== bitmapData.length - 1) bitmapString += '", ';
      }
      bitmapString += '"'; // Add closing quotation mark
      
      document.getElementById('bitmapData').value = bitmapString;
      this.submit();
    });
  </script>
</body>
</html>

```

Jimmy106-e, 2024-03-19:
> @Jimmy106-e can you show me your code please? Also, I refer you to my GitHub repository, you may find the solution there https://github.com/Jumitti/lcd_home_automation

I checked your repo and you have the custom characters already set while mine are set dynamically, from user input.

Jumitti, 2024-03-19:
@Jimmy106-e  Thank you for your answer

I don't understand why you have "0b000" in your error. '''ValueError: invalid literal for int() with base 2: '0b000"00000", "00000", "00000", "00100", "00000", "00000", "00000", "00000"'
'''

Are you sure your HTML code is correct? I am not a specialist and even less in HTML but [here ](https://github.com/Jumitti/lcd_custom_characters )is what I did to create and generate the binary of a custom character (in HTML). And [lightingxd](https://github.com/lightingxd) forked my project into Flask too -> [fork](https://github.com/lightingxd/lcd_custom_characters)

We certainly didn't do any dynamics but I think it's in the creation of the custom character binary that there is a problem

juvus, 2024-04-15:
Hi, sorry for late response. To redefine a custom character you need to pass a python list containing 7 strings as it is shown here: 
`cc.char_1_data = ["01110",
                  "01010",
                  "01110",
                  "00100",
                  "11111",
                  "11111",
                  "11111",
                  "11111"]`
In you case you have 1 string '0b000"00000", "00000", "00000", "00100", "00000", "00000", "00000", "00000"'. And this is not correct stuff causing a problem. 
Solution: write you own parcer (or convertor) from your string (comming from flask app form) to a python list of strings.

Jimmy106-e, 2024-04-22:
ok thanks

cgomesu, 2024-04-27:
thanks for the feedback, @juvus . closing this issue as a completed.

closed 2024-04-27 by cgomesu