the-raspberry-pi-guy/lcd #75: Change GPIO PIN 3

tony-nz, 2024-06-22:
Hey there, is it possible to change the GPIO 3 pin requirement to another GPIO pin?

I require pin 3 for my power button

cgomesu, 2024-06-22:
hi, @tony-nz . it seems like you have at least two options then:

1. move your power button to another pin if it does not use the i2c bus.
1. use the i2c-0 bus via gpio 0 and 1. if you choose to do so, make sure to read the following docs beforehand:
    - https://pinout.xyz/pinout/i2c
    - https://learn.sparkfun.com/tutorials/raspberry-pi-spi-and-i2c-tutorial/all#i2c-0-on-40-pin-pi-boards
    - https://www.raspberrypi.com/documentation/computers/compute-module.html#i2c-mapping-of-gpio-pins

tony-nz, 2024-06-24:
@cgomesu thank you, I will give this ago :)

closed 2024-06-25 by cgomesu