the-raspberry-pi-guy/lcd #77: How can I use this script outsite the lcd folder

jokerper, 2024-06-24:
**Describe the issue**
Add a clear and concise description of the issue here.
Hello I can only use this script insite the lcd folder. How can I install the drivers file globaly in my raspberry pi 4, so I can write to my lcd anyware in my file structure.
**Demo**
- File: `add the filename here`
- (Optional.) Author: 

**Executed command and associated error**
... # Import necessary libraries for communication and display use
import drivers

# Load the driver and set it to "display"
# If you use something from the driver library use the "display." prefix first
display = drivers.Lcd()

# Main body of code
        # Remember that your sentences can only be 16 characters long!
display.lcd_display_string("    jokerper    ", 1)  # Write line of text to first line of display
display.lcd_display_string("  Demo Pi code  ", 2)  # Write line of text to second line of display
add the executed command here
```
```
add the associated error here
```>>> import drivers
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'drivers'

**Host and software info**
- RPi board version: 4b
- OS version: `add here`
- Python version : python3

**Checklist**
- [ ] I have watched and followed the instructions in the [Youtube tutorial](https://www.youtube.com/watch?v=fR5XhHYzUK0).
- [ ] I have searched [open and closed issues](https://github.com/the-raspberry-pi-guy/lcd/issues) for either an identical or similar bug before reporting this new one.
- [ ] I have used Python 2.7

**Additional context**
Add any other context about the problem here.

cgomesu, 2024-06-25:
please watch the tutorial and then provide us with more information about what you're trying to do exactly. I'd appreciate if you followed the issue template as well. I fixed your original submission, so just edit it with what is missing.

cgomesu, 2024-07-06:
stale

closed 2024-07-06 by cgomesu