the-raspberry-pi-guy/lcd #81: Winstar WEH001602A Ecran Oled 16x2

shinzuka, 2025-03-09:
hello,

i have this lcd model
https://www.audiophonics.fr/fr/ecrans-afficheurs-vu-metres/winstar-weh001602a-ecran-oled-16x2-jaune-hd44780-p-10844.html?srsltid=AfmBOooNTRqSM72iN0YE6hp7yvtXbZZLhR04RaJNFdDQmSPvFksplJJO

i will add an i2c module to it will it be compatible with your script?

it is for use with the latest moode audio

cgomesu, 2025-03-11:
if you're doing this for fun/learning, then sure, go ahead and give it a try. otherwise, it's not worth the trouble buying separate components and soldering them.

closed 2025-03-11 by cgomesu

shinzuka, 2025-03-11:
If I do it it's because I already have the Winstar LCD and I can't find any script to make it work in parallel mode.