the-raspberry-pi-guy/lcd #83: does not work on reboot

shinzuka, 2025-03-11:
I must be getting closer to the goal since it appears correctly with the command python3 /home/${USER}/lcd/.

It doesn't work when restarting, there's an error but I can't fix it.

$ systemctl status rpi-lcd.service
● rpi-lcd.service - RPi Python script for a 16x2 LCD
     Loaded: loaded (/lib/systemd/system/rpi-lcd.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2025-03-11 21:21:02 CET; 3s ago
    Process: 22412 ExecStart=/usr/bin/python3 home/${USER}/lcd/lcd.py (code=exited, status=1/FAILURE)
   Main PID: 22412 (code=exited, status=1/FAILURE)
        CPU: 123ms

cgomesu, 2025-03-11:
how does `/lib/systemd/system/rpi-lcd.service` look like? paste the output of the following command:

```sh
cat /lib/systemd/system/rpi-lcd.service
```

from your previous message, it looks like exec is not pointing to a valid executable. substitute `${USER}` for your actual user or set the working directory and run relative to it. many options.

shinzuka, 2025-03-11:
with the username

systemctl status rpi-lcd.service
● rpi-lcd.service - RPi Python script for a 16x2 LCD
     Loaded: loaded (/lib/systemd/system/rpi-lcd.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2025-03-11 21:56:02 CET; 3s ago
    Process: 63940 ExecStart=/usr/bin/python3 home/${shinzuka}/lcd/lcd.py (code=exited, status=2)
   Main PID: 63940 (code=exited, status=2)
        CPU: 49ms

cgomesu, 2025-03-11:
just your username, without ${}

shinzuka, 2025-03-12:
always an error

~/lcd $ systemctl status rpi-lcd.service
● rpi-lcd.service - RPi Python script for a 20x40 LCD
     Loaded: loaded (/lib/systemd/system/rpi-lcd.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2025-03-12 08:14:15 CET; 2s ago
    Process: 6315 ExecStart=/usr/bin/python3 /home/shinzuka/lcd/lcd.py (code=exited, status=1/FAILURE)
   Main PID: 6315 (code=exited, status=1/FAILURE)
        CPU: 123ms
shinzuka@moode:~/lcd

shinzuka, 2025-03-12:
I deleted user=shinzuka and it works. However, he writes anything for the track names. Like "what we do" he writes "that we do". For all the titles he replaces the first letter with a "T"

cgomesu, 2025-03-16:
that sounds great. can i close this issue now that you found a solution?

shinzuka, 2025-03-16:
below is the file to modify

import drivers
import subprocess
from datetime import datetime
import time
display = drivers.Lcd()

#fixed fields
nowstr=time.strftime('%X %x %Z')
hostname="192.168.1.3"

#Load the values ​​from the current song file
songattr={}
with open("/var/local/www/currentsong.txt") as file1:
 for line in file1:
 name, var = line.partition("=")[::2]
 songattr[name.strip()] = var.strip()

display.lcd_display_string(songattr["artist"], 1)
display.lcd_display_string(songattr["album"], 2)
display.lcd_display_string(songattr["title"], 3)
#display.lcd_display_string(songattr["date"], 4)
#display.lcd_display_string(hostname, 4)
display.lcd_display_string(songattr["track"], 4)

I would like to make the following changes.

- have the track number on the same line as the song title.
- on the last line, the characteristics of the file being played (e.g., 24-bit 44.1 kHz).
- set up a sleep mode if not playing, with the time and IP address of the Moode interface displayed.

What I have so far

[image]

shinzuka, 2025-03-21:
I'm making a bit of progress by looking for information on the internet.
[image]

closed 2025-03-23 by shinzuka