the-raspberry-pi-guy/lcd #84: LCD_16x2 package

Jumitti, 2025-03-26:
Hello everyone,

I saw in the closed Issue (#80) a person asking for a package for LCD.

Well I tried to do my best, I did an LCD_16x2 0.1.0-Alpha1 version here [REPO PACKAGE](https://github.com/Jumitti/lcd_home_automation/tree/package)

You can install it (and especially try it to find the bugs):
- With GitHub:
```bash
pip install git+https://github.com/Jumitti/lcd_home_automation.git@package
```
- With Pypi:
```bash
pip install LCD-16x2==0.1.0a1
```

Small tips etc... :
- use``from lcd_16x2 import drivers`` instead of ``import drivers``: pourquoi ce changement ? I developed an emulator PR #79  [LCD Emulator](https://github.com/Jumitti/lcd_home_automation/tree/simulator) So you might as well integrate it now
- Same if you use the emulator: ``from lcd_16x2 import emulators`` instead of ``import emulators``
- If you test demos: ``python -m demo.demo_clock`` (change demo_clock by the demo available in the folder [demo](https://github.com/Jumitti/lcd_home_automation/tree/package/demo))
- There is a OS recognition. If you are on Raspeberry, then the package will require the necessary to install what we were doing with ``sudo./Install.sh``. If you are on Windows or MacOS, since the 'smbus' specific packages in Raspberry (the pines and connectors) are not, then it will also be necessary so that you can use the emulator anyway

I would do other updates if I forgot things, I greatly invite you to try and make feedback...

Jumitti, 2025-03-26:
So sorry, I didn't have the faith tonight to reinstall an OS and start from scratch on my Raspberry. So I just tested ``pip install LCD-16x2`` and a script I already had and everything works fine using ``from lcd_16x2 import drivers``.

However, my OS recognition has a small problem

cgomesu, 2025-03-26:
k. thanks for the updates, @Jumitti . I'll be quite busy over the next few days, so it might take a while before I can get back to you re the changes you proposed. just a heads up.

Jumitti, 2025-03-26:
Yes, no worries, it will be the same for me... Especially since in the end, it works quite well for drivers and emulators but install.sh is buggy so I'll have to investigate a little more
Anyway it's mainly to say that I'm working on it, we'll see how long it takes to get it
Thank you again for always being there @cgomesu